sinara-hw/Kasli-SOC #1: Zynq Kasli

dtcallcock, 2019-12-02:
A new Kasli based on the forthcoming v1.2 design, but with a Zynq-7000 series FPGA (2x ARM CPU) instead of an Artix-7 (no hard CPU).

This will allow the proposed [Zynq version of Artiq](https://github.com/m-labs/artiq/issues/1167) to utilise the EEM hardware.  

The proposed FPGA is the XC7Z030-3FFG676I.  The reasons for choosing this FPGA (as laid out by @hartytp) are:
- Very similar (same Kintex-7 fabric) to the XC7Z045 FFG900 –2 used on the [Xilinx ZX706 Evaluation Kit](https://www.xilinx.com/products/boards-and-kits/ek-z7-zc706-g.html) that will likely be used for gateware development.
- The CPUs on the 7Z030 go up to maximum 1GHz clock rate available on 7000 series Zynqs.
- Fastest speed grade to ensure we can clock the AXI bus as fast as possible. The thought is that cost sensitive users can stick with the Artix Kasli, and the main user base for Zynq Kasli will be people who don’t mind paying an extra $50 to get the fastest FPGA.
- The design costs + risks will be much lower if we use a Zync model that @gkasprow has used in previous designs. He's used this FPGA before.
- It has enough pins to drive all 12 EEM connectors + WR oscillators

SDRAM would be routed to the ARM subsystem only.  The gateware will still have DMA access based on tests carried out by @cjbe, where he found that:

> I have tested DMA from PL to PS-SDRAM via the HP ports, and indeed it works without any problems, and one gets the advertised bandwidth.

I am working on getting this board funded and would appreciate any input.

Also, is 'Zynq Kasli' a confusing name?  Do we need to dust off our Russian maps and come up with something more unique?

gkasprow, 2019-12-08:
My student will take care of this board very soon. He will post new issues so don't be surprised.

hartytp, 2019-12-29:
XC7Z030-3FFG676 sounds like a good choice for this.

As mentioned before, I'd ideally like to go for 4 SPF + 1 RJ45 on the front panel, copying ethernet from the zc706.

Ram to the PS only.

Other than that, stick as close to the current Kasli 2.0 design as possible.

gkasprow, 2020-01-06:
The schematics are close to completion. 
Btw, @sbourdeauducq did you see [this](https://www.electronicproducts.com/Digital_ICs/SoCs_ASICs_ASSPs_MEMS/Microchip_offers_early_access_to_PolarFire_SoC_FPGA.aspx?utm_source=newsletter&utm_campaign=link&utm_medium=InspiringDesign-20200106)

sbourdeauducq, 2020-05-05:
>     * no SD Card

We want a SD card if possible. From experience on ZC706, it might be the least troublesome way to get non-volatile storage on Zynq, and there's the easy "escape hatch" of extracting the SD card and putting it into a $5 USB adapter in case Zynq won't write to it. The PS NOR flash controller is messed up; the only thing that seems to work semi-reliably is a read-only mode where the flash contents are mapped into the CPU memory. We didn't manage to write to the flash at all. There may be some workarounds in some cases like re-routing the flash to GPIO and bit-banging, but it's extra trouble and development time.

filipswit, 2020-05-07:
I'll try to fit it. It looks there is enough space above RAM but only for this kind of socket: https://www.molex.com/molex/products/part-detail/memory_card_socket/0473092651

The white line is the outline of microSD
[image]
[image]

gkasprow, 2020-05-07:
I used a slightly higher version of this Molex connector and managed to fit the BGA chip under the SD card. You can use it as well and rotate the connector if that helps.

sbourdeauducq, 2020-05-08:
How well does this connector hold the SD card?
We'll want to ship systems with the SD card pre-installed, and carriers sometimes handle packages roughly - while they have teams whose job is to find excuses to deny claims. And just normal transportation vibrations tend to dislodge things. So our hardware should be pretty sturdy.

gkasprow, 2020-05-09:
We used these connectors in the production of 8k pieces of LTE routers. There were several issues with these boards but it never happened that SD slipped off.
Here are some photos
[image]
[image]
[image]

jordens, 2020-05-09:
They have the standard spring loaded ratchet locking mechanism, right? There is no way they'd fall out. Get proper packaging and tie down cables properly and make sure fasteners designs are right and tightened properly.

filipswit, 2020-05-16:
Version with SD card is on a repo.

closed 2020-05-29 by filipswit