sinara-hw/Kasli-SOC #3: Zynq reset/JTAG fiasco

sbourdeauducq, 2020-01-11:
The Zynq reset and JTAG interface is complicated and temperamental, and in particular we observed the following problems:
* It requires a SRST signal controlled by the JTAG adapter. This signal is a bad idea as reset commands could be delivered over JTAG just fine without the various complications associated with an additional signal. This is apparently done by ARM for power consumption reasons (keep JTAG powered down until the first SRST), but a much better design would have been a level-sensitive "JTAG enable" pin that could be tied high with a jumper in hardware. (In Xilinx's defense, this is a mistake that ARM made)
* A pulse on SRST that is too short makes the ARM cores fall of the JTAG chain (they disappear even if they had been detected just before the short pulse), sometimes until the Zynq chip is power cycled.
* Attempting to use JTAG while one of the ARM cores is stopped breaks the JTAG interface, again, sometimes until a power cycle. Note that stopping an ARM core can be done in firmware; in other words, Zynq is designed in a way that using JTAG at the wrong time depending on what the firmware is doing can put JTAG in an unrecoverable state.

There is a "PS_POR_B" input on Zynq that may or may not be used in lieu of a power cycle. [The technical reference manual says](https://www.xilinx.com/support/documentation/user_guides/ug585-Zynq-7000-TRM.pdf) "This signal resets every register in the device capable of being reset". Trust but verify... 

I propose:
* Be very careful about SRST and POR signals on this board. Maybe expose the traces when routing in a way that the board can be easily manually reworked in case of problems.
* Make POR accessible to the debug computer. Putting it on a GPIO expander accessible on one of the I2C buses connected to a FTDI channel sounds like a good idea to me because 1) this FTDI/I2C system worked fine on Kasli 2) we are not exposed to FTDI GPIO problems 3) it makes it harder to accidentally trigger a POR e.g. by using the wrong driver on that FTDI channel.
* Do careful experimentation with POR: put the Zynq into "JTAG broken" states and check that it can always recover with a POR. This can be done on the ZC706 which has POR control on a pushbutton.
* If it turns out that POR doesn't work as documented or if we still don't trust it, also make the Zynq power supply control accessible on I2C.

filipswit, 2020-01-22:
I changed the power sequencing system - https://github.com/sinara-hw/Kasli-SOC/issues/4
So the power supply system should just work.
Now the POR is controlled by supervisor IC and eventually by GPIO. SRST is normally pulled high (unused), optionally you can drive it from GPIO.  Of course during the layout I'll try to put all these traces on external layers.

sbourdeauducq, 2020-01-23:
> SRST is normally pulled high (unused), optionally you can drive it from GPIO.

That's not optional. We **need** to be able to drive it from the JTAG adapter, and to avoid further complications, the SRST layout and polarity must match those of existing common JTAG adapters.

filipswit, 2020-01-26:
Ok. Default mounting will allow it.

closed 2020-05-29 by filipswit