sinara-hw/Kasli-SOC #23: SDRAM reference?

sbourdeauducq, 2020-05-10:
Is the SDRAM configuration an exact copy of a Xilinx development kit, and which one?
Getting the PS SDRAM controller to work is rather difficult, so it would be good to simply reuse something that is known to work (also we could simply run the existing FSBL for a quick test of the board) and for which the correct PS register values can be looked up.

filipswit, 2020-05-15:
It's connected without any pin swapping.

sbourdeauducq, 2020-05-16:
What about the SDRAM chip P/N including timing, number of rows/columns, etc.

filipswit, 2020-05-16:
I've used: MT41K256M16TW-107:P

sbourdeauducq, 2020-05-17:
Is there a Zynq Xilinx kit with this exact chip?

filipswit, 2020-05-20:
I've found only some chinese kits: qmtech zynq7000 and Lychee HEX ZYNQ702.

sbourdeauducq, 2020-05-21:
Could we use a SDRAM system that is an exact copy of a better-known kit? Is there a reason not to?

sbourdeauducq, 2020-05-21:
For example, Cora Z7-10. Not that we have managed to get the SDRAM to behave itself on this board without the Xilinx junk libraries (we have only done so on the ZC706, on the Cora there is typical Zynq behavior) - but at least we can use the latter as a workaround.

filipswit, 2020-05-21:
Well, SDRAM from Cora -  MT41K256M16HA-125:E looks fine for me.

sbourdeauducq, 2020-05-23:
OK.

closed 2020-05-23 by sbourdeauducq