sinara-hw/Kasli-SOC #25: RAM: Copy KZ706 exactly

hartytp, 2020-08-06:
Please can we exactly copy the ram from the KZ706 board?

filipswit, 2020-08-07:
The exact copy is unbuyable. Current chip is used in Cora Z7-10 - #23

hartytp, 2020-08-07:
Thanks @filipswit. "You can't buy it" is a pretty good reason not to use the chip on the ZC706. What's the most similar replacement? I have a preference for lots of ram with a nice wide bus here... @sbourdeauducq thoughts?

filipswit, 2020-08-08:
In general this memory family looks to be little available: https://pl.mouser.com/Search/Refine?Keyword=MT41J256M8&Ns=Access+Time%7C1

The family in FBGA-96 package looks to be more popular: https://pl.mouser.com/_/?keyword=MT41K256M16. We can easily use DDR3 or DDR3L chips. 

Another thing why I don't like this idea is that they are in different package so the RAM will need to be rerouted.

hartytp, 2020-08-10:
@sbourdeauducq what's your preference here? Again, reiterating my preference for a large/wide RAM on Kasli-Soc...

sbourdeauducq, 2020-08-10:
The RAM on Kasli-SoC is 32-bit no matter what - or did I miss out something?
If the ZC706 RAM cannot be purchased I don't think there's a better option than "try the RAM that's on Cora-Z7, but 32-bit instead of 16-bit, and see what happens".

sbourdeauducq, 2020-08-10:
Does the memory capacity at the level of commercial DRAMs (hundreds of MB) really matter?

hartytp, 2020-08-10:
> Does the memory capacity at the level of commercial DRAMs (hundreds of MB) really matter?

IIRC @dnadlinger has had issues overflowing the memory on Kasli

dhslichter, 2020-08-11:
> IIRC @dnadlinger has had issues overflowing the memory on Kasli

Ouch!  How?

dnadlinger, 2020-08-11:
> Ouch! How?

There was a memory leak at some point, but the bigger issue was with DMA, which currently panics the comms CPU if you run out of space, which can happen quite easily as sequences persist across experiments (and we currently don't have an API to just erase all of them in cases where persistence is not wanted).

I haven't checked how much RAM we are actually using at any one time, or done the math on how that number would change for e.g. long transport sequences with a few Fastinos.

sbourdeauducq, 2021-05-28:
RAM working correctly so far on prototype.

closed 2021-05-28 by sbourdeauducq