sinara-hw/Kasli-SOC #26: Ethernet: copy KZ706

hartytp, 2020-08-06:
Please can we make sure that the ethernet is an exact copy of the KZ706?

filipswit, 2020-08-07:
Ok, I may change it to 88E1116.

filipswit, 2020-08-08:
How should I set configuration pins Config[0..3]?

hartytp, 2020-08-10:
^ @sbourdeauducq

sbourdeauducq, 2020-08-10:
The default jumper settings for ZC706 work here. Beyond that I don't particularly know.

gkasprow, 2020-08-12:
what's wrong with 88E1512? it's already used in Sayma and Marble board and it works with Litex.

sbourdeauducq, 2020-08-13:
It's probably fine, just not tested with Zynq and AFAIK not tested on Sayma either (IIRC this was replaced on v2.0, and never used there, though this will probably change soon since we're still having problems with the GTH)...
To put it politely, the LiteX policy is to be optimistic about what "works" with it. GTH Ethernet also "works" with LiteX.

sbourdeauducq, 2020-08-13:
Does it need any MDIO transactions before it operates or does it power up with functional default settings?

gkasprow, 2020-08-13:
Yes, AFAIR it needs one transaction to set up the interface to RJ45 mode. It's quite a new chip and supports several modes like IEEE1588 and SyncE. It's used in several dev boards as well. And it has an official open datasheet. The 88E1116 is old and power-hungry.
It also supports SFP modules directly. But I do not insist very much on this solution.

sbourdeauducq, 2020-08-13:
Okay, we have code for MDIO on Zynq but I need to check if it works, which can be done on ZC706. As you have seen with I2C even simple PS peripherals can be troublesome...

sbourdeauducq, 2020-08-14:
It works. So, in theory, adding the 88E1512 init should be inexpensive. Please still connect all Ethernet signals to the PS directly, though.

gkasprow, 2020-08-14:
yes, PS connectivity is good idea. Use same pins as on the devkit. Then it will work without the bitstream. If we want PL access to Ethernet, will use transceivers and SFP.

filipswit, 2020-08-14:
And revert 88E1512, right?

gkasprow, 2020-08-14:
@filipswit do what is easier for you, in terms of routing

sbourdeauducq, 2020-08-15:
>  Then it will work without the bitstream

And we won't have to worry about meeting timing on RGMII.

filipswit, 2020-09-20:
I placed 88E1512 and connected it directly to PS.

closed 2021-05-28 by sbourdeauducq