sinara-hw/Kasli-SOC #27: Align clocking tree to DI/OT

jordens, 2020-08-11:
https://github.com/sinara-hw/meta/issues/56#issuecomment-670604266

filipswit, 2020-08-15:
@gkasprow how about this? Do we make changes in clocking in this revision or in future?

hartytp, 2020-08-18:
@jordens AFAICT (but correct me if I'm wrong) the only part of that post that's relevant here is this bit

> Clocking: IMO Kasli-SoC should use the DI/OT clocking tree (add 4 MMCX). It support all options (WR CR with VCXO, DCXO, and Si5x-based recovery).

Could you sketch out (phone snap of a paper sketch completely fine) what solution you want (eliminates room for confusion between people comparing schematics manually). 

> I'd like profit from the CERN involvement (review, development, characterization, testing, volume, cost). Pretty sure that Kasli-SoC will not profit much from that.

If I'm reading you correctly, the motivation for these changes on the DI/OT card was to make something that's compatible with both ARTIQ and CERN so that we can benefit from CERN's involvement. That makes sense in that context, but CERN are not going to use Kasli-SoC whatever we do with the clocking, so the motivation for applying these changes here is unclear to me.

Why not stick with the clock tree on Kasli v2.0 which (other than the WR) is known to work for our purposes? Why add other components which don't have any clear users?

Could you articulate the benefit from implementing these changes on Kasli-SoC as opposed to keeping Kasli-SoC as close as possible to "Kasli v2.0 with Zynq" and converging on the CERN designs in the DI/OT board?