sinara-hw/Kasli-SOC #29: JTAG and reset validation

sbourdeauducq, 2020-09-03:
* identify or write an OpenOCD cable file that can be used with the FTDI chip as connected on that board.
* test that the POR GPIO is actually working with the FTDI chip configured as intended on this board (JTAG, I2C, UART).
* test that toggling the POR or SRST GPIO does not break JTAG, UART, or I2C. FTDI GPIO controllers are extremely buggy and there are lots of surprises like that.
* test that using OpenOCD with the cable files does not trigger POR.
* test that plugging the USB cable with the default (UART) driver attaching does not trigger POR.

This should be done before board production.

hartytp, 2020-09-03:
@sbourdeauducq to check I understand the situation here...

The ZC706 has two JTAG headers and a [digilent USB JTAG module](https://store.digilentinc.com/jtag-smt2-surface-mount-programming-module/). Am I right in thinking that:
- you started using the digilent USB JTAG module but didn't get that to work
- so you switched to using an Olimex USB JTAG adapter going into one of the headers
- the Olimex programmer works reliably but is less convenient than the USB solution

The current Kasli-SoC schematic has a FT4232H (like Kasli-Artix) for JTAG instead of the digilent module as well as a standard 2x7 header.

Am I right in thinking that the 2x7 header combined with the Olimex cable work reliably since it's the same configuration you've already tested? So, we have that as a "known good" fall back?

Is your concern that (due to implementation of the Zynq JTAG interface) the FTDI chip might be a pain to get working? Or are you worried that we might just not be able to get it to work at all?

Are you suggesting that we should get this working prior to sending the prototypes off to manufacture? If so, what hardware configuration do you want to see used in the tests? Or, since we have the Olimex adapter as fallback, are you suggesting we keep the hardware as is for now, but make sure to allocate resources for testing/development once the hardware arrives?

hartytp, 2020-09-03:
e.g. are you suggesting that someone should take a FT4232H eval module and wire it up to the JTAG header on a KC706 at this stage?

It sounds like the POR/GPIO tests you want to see can all be done using a FT4232H eval board only, right?

sbourdeauducq, 2020-09-03:
> The ZC706 has two JTAG headers and a digilent USB JTAG module. Am I right in thinking that:
>
>    you started using the digilent USB JTAG module but didn't get that to work
>    so you switched to using [...]

Correct.

>  are you suggesting that someone should take a FT4232H eval module and wire it up to the JTAG header on a KC706 at this stage?

That would be a very good thing to do, yes. Maybe also POR with a bit of soldering.

hartytp, 2020-09-03:
@gkasprow what do you think about this? Could one of your students wire up a FT4232H eval module to a Zynq evaluation board like a ZC706 and confirm that they can flash the FPGA from openocd?

sbourdeauducq, 2020-09-03:
Also, we should try to use a layout that is compatible with Vivado. And also check that Vivado won't toggle POR. But that's low importance.

sbourdeauducq, 2020-09-03:
>  confirm that they can flash the FPGA from openocd?

Run it a couple hundred times, many Zynq JTAG bugs are intermittent.

hartytp, 2020-09-03:
> Run it a couple hundred times, many Zynq JTAG bugs are intermittent.

Ok. Your initial issue description didn't make it clear what test you wanted (hardware configuration, what should be done, etc) and what the criteria for success is. So, we're talking about having a loop that repeatedly flashes a binary via openocd without openocd raising an error?

gkasprow, 2020-09-03:
That's interesting. I didn't manage to enter the debug session with a standard red Xilinx programmer, while Vivado worked fine with it. 
HS2 from Digilent works fine in both cases. However, I tried it with ZynQ US+.
ZC706 uses also Digilent module based on FT2232
HS2 is just FT2232. The only difference is the EEPROM content.

gkasprow, 2020-09-03:
Yet another option could be using direct access to the PS JTAG over MIO pins. FT4232 has two JTAG controllers that could be run independently

sbourdeauducq, 2020-09-04:
> That's interesting. I didn't manage to enter the debug session with a standard red Xilinx programmer,

There's plenty of "interesting" behavior like that with JTAG on Zynq, which is why we must be careful.

> However, I tried it with ZynQ US+.

US+ has a different ARM core and JTAG system, so tests on US+ are mostly irrelevant to Kasli-SoC.

> ZC706 uses also Digilent module based on FT2232
> HS2 is just FT2232. The only difference is the EEPROM content.

Right. That's why it may be interesting to try to stay compatible with Vivado on Kasli-SoC.

> Yet another option could be using direct access to the PS JTAG over MIO pins.

That second JTAG port is enabled only after software has turned it on, so it creates a chicken-and-egg situation. As far as I know, Xilinx added that second port only because a number of electronics engineers love to use legacy debuggers for ARM that break if there's something else on the chain. This also means we could not use that port to debug the PL. And there is no indication that it would work better than the main JTAG that has PS+PL.

sbourdeauducq, 2020-09-07:
One data point: on the Ultra96 JTAG board (AES-ACC-U96-JTAG), Avnet are using a FT2232H with both SRST and POR on GPIOs. So it seems possible to do that. Maybe we can try that JTAG board with OpenOCD and the ZC706 and see if it works.

sbourdeauducq, 2020-09-07:
Gave it a quick try, got the standard fare "Error: JTAG scan chain interrogation failed: all ones", so someone does need to spend a week on that thing.

sbourdeauducq, 2020-09-07:
Also, plugging the USB connector (without running anything on the computer) into the AES-ACC-U96-JTAG resets the Zynq, so this pod is an example of what not to do...

closed 2021-06-21 by sbourdeauducq