sinara-hw/Kasli-SOC #31: compare with kasli 2.1

hartytp, 2020-11-10:
Let's check that all relevant changes from Kasli >= 2.0 have been included in this

gkasprow, 2021-01-05:
I recently added 6pins spaced by 2.54mm to support 4 direct IOs

filipswit, 2021-01-06:
Ok, I can fit additional 4 IOs. 
I've noticed that SMA input is biased to 1.25V after balun instead of AC coupled. 
[image] 
[image]

gkasprow, 2021-01-06:
I already added the IOs. The FPGA actually needs DC coupling and it's good to have external 100Ohm matching.

filipswit, 2021-01-06:
But last commit is mine and is without it.

gkasprow, 2021-01-06:
I didn't do the commit yet so don't modify your project

filipswit, 2021-01-06:
I change fan connector to be with key. 

Should I add capacitors before and after EMI filter FL*?