sinara-hw/Kasli-SOC #52: Fan connector too close to PCB edge

kaolpr, 2021-05-17:
When fan connector is connected PCB can not be placed in the 3U crate because of too small distance from the PCB edge. For the prototype it can easily be solved by replacing fan connector, however for batch production it would be more efficient to move connector a bit towards the PCB center.

[image]

sbourdeauducq, 2022-01-27:
Also the fan polarity is wrong.

gkasprow, 2022-01-27:
there is no single standard of fan pinout :)

sbourdeauducq, 2022-01-27:
Even with the same P/N from Radiant heatsinks?
The picture above also shows the wrong polarity; the red wire should be facing towards the back of the card.

gkasprow, 2022-01-27:
oh, you are right.