sinara-hw/Kasli-SOC #53: Ethernet PHY reset may need to be asserted during power up

sbourdeauducq, 2021-05-28:
The Zynq cannot do it. Not sure if another reset after power-up is sufficient. Currently (without any reset) Ethernet communications work but not MDIO.
[image]

sbourdeauducq, 2021-05-28:
On the ZC706 (with a similar but not identical PHY) they have a gate to generate the power-up reset, so this could be important.
[image]

sbourdeauducq, 2021-05-29:
corresponding firmware issue: https://git.m-labs.hk/M-Labs/zynq-rs/issues/78

sbourdeauducq, 2021-07-05:
Having or not having a power-up reset does not seem to make a difference, PHY works in both cases.
Tested by @TopQuark12

TopQuark12, 2021-07-05:
Current hardware configuration works fine as is, holding rst down during power on does not affect the behavior of the phy chip in observation. Would be helpful if the schematics indicate the PHY MDIO address is 0.

gkasprow, 2022-04-21:
added note

closed 2022-04-21 by gkasprow