sinara-hw/Kasli-SOC #54: install J24 jumper (USB SRST) during production

sbourdeauducq, 2021-05-29:
Confirmed working (with multimeter and effect on Zynq) using:

```
ftdi_layout_init 0x0098 0x008b
...
ftdi_layout_signal nSRST -oe 0x0004
reset_config srst_only srst_open_drain
adapter_nsrst_assert_width 250
adapter_nsrst_delay 400
```

This can be tested easily with the ``ftdi_set_signal nSRST 0/z`` command in OpenOCD.

We'll probably want the POR jumper as well, though I have not tested it yet.

BTW is there a FTDI quirkologist here who knows why need those values 0x0098 0x008b in ftdi_layout_init? I copied them from another cable, and changing them typically breaks JTAG.

sbourdeauducq, 2021-05-29:
Got POR to work as well and similarly tested. I had to use PyFTDI, no dice with OpenOCD; but it doesn't matter - we can just integrate the PyFTDI program into the CI/dev scripts.

filipswit, 2022-11-15:
added

closed 2022-11-15 by filipswit