sinara-hw/Kasli-SOC #55: master issue: Kasli-SoC v1.0 testing

sbourdeauducq, 2021-05-29:
This Issue will be edited to reflect the current status of the v1.0 hardware functional testing.

* internal JTAG: OK
* nasty Zynq programming signals (POR and SRST): OK, needs jumpers (#54)
* external JTAG: OK (tested with Digilent HS2)
* UART: OK
* SDRAM: OK
* SD card access from firmware: OK
* SD card boot: OK
* I2C: does not work, https://git.m-labs.hk/M-Labs/artiq-zynq/issues/130
* Ethernet: data OK, MDIO does not work (#53 / https://git.m-labs.hk/M-Labs/zynq-rs/issues/78)
* WRPLL: cannot test, Si549 not available
* EEM slots: not tested
* SFPs: not tested
* Fan control: not tested
* External clock input: not tested
* LEDs: ERROR LED OK, others not tested

sbourdeauducq, 2021-05-30:
Built a system with Kasli-SoC + 12 DIO cards in output mode, and ran artiq_sinara_tester to test EEM slots. All passed.
[image]

sbourdeauducq, 2021-06-19:
Excessive voltage on the T21 gate is holding the IC14/IC15 RESET asserted.
I2C waveforms on I2C_SW_1V8 and 3V3 are otherwise looking ok.

sbourdeauducq, 2021-06-19:
The voltage is around 800mV and those MOSFETs have a particularly low threshold (400mV). It seems the pulldown R33 is not sufficient to compensate for the leakage current through the unconfigured Zynq MIO pin.

sbourdeauducq, 2021-06-19:
This can be fixed in firmware by simply configuring the MIO pin:
```
          slcr.mio_pin_33.write(
                slcr::MioPin33::zeroed()
                    .l3_sel(0b000)
                    .io_type(slcr::IoBufferType::Lvcmos33)
                    .pullup(false)
                    .disable_rcvr(true)
            );
```
It will break the FTDI direct I2C access on the other hand.

But the I2C switches are still not responding even though RESET is now deasserted.

sbourdeauducq, 2021-06-19:
NACK on 0x70 is due to a software bug (SCL/SDA race).
0x71 continues to NACK for unknown reasons.

sbourdeauducq, 2021-06-19:
Probing the 3V3 I2C bus.

Writing to IC14, which works:
[image]

Writing to IC15, which does not:
[image]

gkasprow, 2021-06-21:
Xilinx FPGAs have a PUDC pin that defines pullups bahaviour. I don't know if this affects MIO pins as well

sbourdeauducq, 2021-07-05:
PHY MDIO working, was essentially a software problem (found by @TopQuark12)

gkasprow, 2022-04-21:
nothing to do here, closing