sinara-hw/Kasli-SOC #59: Stock problems

pkozakiewicz, 2021-08-31:
I would like to address numerous stock problems regarding this project:

NC7WZ07P6X
TPS7A7200RGW
TPS7A7100RGW
88E1512-XX-NNP2I000
DSC1001DI5-025.0000T
7M-12.000MAAJ-T

I would like to create a clear list of alternatives that could be placed instead.
@gkasprow could you please help?

Programmable oscillators and FPGAs are also unavailable at the moment, but there's nothing that we can replace them with.

gkasprow, 2021-08-31:
Was the PCB produced already?

gkasprow, 2021-08-31:
NC7WZ07P6X -> SN74LVC2G07DCKR;
TPS7A7200RGW needs PCB changes
TPS7A7100RGW needs PCB changes
88E1512-XX-NNP2I000 -> any 88E1512 would work with same package
DSC1001DI5-025.0000T -> any 2*2.5 1.8V 25MHz 15ppm oscillator, any of [them ](https://pl.mouser.com/Passive-Components/Frequency-Control-Timing-Devices/Oscillators/_/N-7jdfiZ1yzvvqx?P=1z0wnusZ1yzvlhvZ1y9ho36Z1yzt1v4Z1yzttetZ1z0vxlvZ1y9k47pZ1z0jlkxZ1z0w7wnZ1z0w6riZ1yzn4qaZ1z0waa1Z1z0s1xeZ1z0we1tZ1z0s1x7)should work; use TG2520SMN 25.0000M-ECGNNM3 or ASD3-25.000MHZ-LR-T
7M-12.000MAAJ-T -> any 3.2x2.5, 4 pin, 12MHz, <50ppm, 18pf crystal would work; ABM8-12.000MHZ-R100-D7G-T, ECS-120-18-33-AGN-TR,830069944

pkozakiewicz, 2021-09-01:
> Was the PCB produced already?

Yes, but we can consider rejecting it or using it later. I guess that placing double footprints wherever possible would be a good practice. AFAIK the market situation is unlikely to improve by the end of 2022.

pkozakiewicz, 2021-09-01:
> TPS7A7200RGW needs PCB changes
> TPS7A7100RGW needs PCB changes

These are available, but their current price is much higher than PCB tooling and manufacturing cost.

closed 2022-04-21 by gkasprow