sinara-hw/Kasli-SOC #63: Question about drill holes for the front panel

HarryMakes, 2021-12-16:
We found that the drill holes for assembling the front panel look inconsistent with other Sinara PCBs in the following ways:

#### Kasli-SoC v1.0rc4
[image]
#### IDC-SMA v1.0rc1
[image]

1. The copper pad for each hole has an outer diameter of 2.81 mm. However, it is 5.00 mm on other PCBs such as Urukul and IDC-SMA.
2. The solder mask on these locations has the same diameter of 2.81 mm. However, it is 5.12 mm on other PCBs, which is slightly larger than the copper pad.
3. Using the Altium online viewer, the distance (presumably from centre to centre) between the two holes is 87.63 mm on Kasli-SoC. However, it is 83.90 mm on other PCBs. 

For (1), is it an error that the copper pad is almost identically-sized as the drill hole (2.8 mm in diameter)?

For (2), does it matter for the solder mask to be slightly bigger than the copper pad around each of these holes? Can they be of the exact same diameter as the copper pad, i.e. 5.00 mm?

For (3), has the distance changed for Kasli-SoC? On Altium it seems that the "component" for these holes are different from other PCBs - on other PCBs they are known as B1 and B2 "plated holes", but on Kasli-SoC they don't have designators. On other PCBs they are usually connected to GND but not on Kasli-SoC. With the current Kasli-SoC v1.0rc4 drilling data, will the normal 3U 8HP front panel not fit on the PCB?

Thank you!

HarryMakes, 2021-12-16:
CC @sbourdeauducq

gkasprow, 2021-12-16:
This looks like a question from the PCB house. So you are actually making the Sinara HW in China/TW...
The Sinara community should be aware of this.

gkasprow, 2022-04-21:
fixed in v1.0.1

closed 2022-04-21 by gkasprow