sinara-hw/Kasli-SOC #65: v1.1

jordens, 2022-02-01:
@gkasprow @sbourdeauducq @hartytp and whoever else is interested

I'm not fully aware of what the current state of Kasli-SoC is beyond what I can tell from the repositories. We are being asked regularly what the status and are mostly recommending that people wait or get involved actively. Two things:

* Is Kasli-SoC being used somewhere actively and more or less continuously or is everybody waiting for somebody else to go through the v1.0 hardware pain phase?
* Does it make sense to address the current HW issues in a v1.1 now (or is it good enough/not enough issues to bother)?

hartytp, 2022-02-01:
Been trying to get our hands on one for a very very long time, but the ship date from CTI keeps slipping...

gkasprow, 2022-02-01:
Yes, the lead time of many ICs is a serious issue. These days I don't even start a new project before making sure all ICs arrived.

jordens, 2022-02-01:
@pkozakiewicz apparently has the parts and could make some, waiting for a decision regarding v1.0 vs v1.1 boards.
We are in the queue for a couple but happy to reduce if someone wants to forge ahead.

gkasprow, 2022-02-01:
I'd suggest fixing only minor issues for the moment. I can assign a dedicated engineer in 6 weeks who would implement all changes.

dhslichter, 2022-02-01:
@jordens for those interested in a Zynq core device with Sinara hardware on a shorter timescale, one option is to use ZC706 as a DRTIO root node and then connect to satellite Kasli(s) in Sinara crate(s) via optical fiber. NIST has funded DRTIO master/satellite for ZC706 and KC705, and we are debugging ZC706 master/Kasli satellite on the magtrap setup at present.

jordens, 2022-02-01:
I am aware but not interested in the ZC706. This is about Kasli-SoC.

hartytp, 2022-02-01:
@cjbe will know the details better than me, but IIRC we ordered a pretty large batch from CTI a while ago and don't plan to order more until testing has been done. CTI are still having stock issues (even if TS have some).

sbourdeauducq, 2022-02-02:
No major issues in v1.0 from what I can tell, only what I reported.

pkozakiewicz, 2022-02-03:
@jordens I have one board left that was produced for our internal use. We have finished our work with it. Would you need it for hands-on experience before confirming production of the current release?

pkozakiewicz, 2022-02-09:
What is the final decision on this topic? Shall we wait for v1.1 or produce v1.0 immediately?

gkasprow, 2022-02-11:
@pkozakiewicz Produce v1.0 but make R33=2k2

pkozakiewicz, 2022-02-14:
@gkasprow @jordens @sbourdeauducq @hartytp Good, we are starting production.

gkasprow, 2022-04-21:
fixed in v1.0.1

closed 2022-04-21 by gkasprow