sinara-hw/Kasli-SOC #68: LDO instable - input capacitance to low

mkiepiela, 2022-07-26:
Input capacitance at IC35 is too low (only C125 = 1 uF). It causes voltage instability both, at the output (1V0) and input (1V8) side. As an effect MGTs transceivers may not work correctly.

[image]

[image]

After adding 2.2 uF capacitor in parallel to C125, supply voltage is correct and MGTs work well.
Basically, all LDOs should be reviewed, as there are more LDOs with 1 uF input capacitance. Probably some of them also should be increased.

gkasprow, 2022-07-26:
Good catch.
Moreover, take into account C=f(V) effect in ceramic capacitors. Real capacitance is often twice lower than rated.

nkrackow, 2022-08-02:
So I think the other affected caps are C118, C113 and C130? Or are there more at other places that I might have missed?

mkiepiela, 2022-08-03:
Yes, it looks like there are only these 3 caps more as you pointed above. However I looked into datasheet of LDO and it suggests using 1 uF input cap in combination with low line impedance.

[image]

So far we encountered the problem only with 1V0 LDO when MGT transceivers were active (high current consumption from 1V0 domain). Other voltages were stable and did not need additional capacitance.
Moreover, there is the least dropout value on this particular LDO (1V8 - 1V0 = 800 mV) what increases the risk of instability.

The next one with low dropout value (1.35 V) is IC33 (1V35 VCCDDR), however we did not observe instability there. But for now we were able to check it only on one board.

Summarizing, I think that adding capacitor only at 1V0 LDO seems to be a good enough solution, unless someone observe similar problem on any other LDO.

nkrackow, 2022-08-03:
Ok, thanks!

filipswit, 2022-11-15:
Added 10uF at inputs for all 4 LDOs