sinara-hw/Kasli-SOC #69: changelog v1.0.2

filipswit, 2022-11-15:
- [ ] fixed https://github.com/sinara-hw/Kasli-SOC/issues/68
- [ ] fixed https://github.com/sinara-hw/Kasli-SOC/issues/67
- [ ] added 24AA025E48T-I/OT and  24AA025E48-I/SN on separate I2C bus https://github.com/sinara-hw/Kasli-SOC/issues/66
- [ ] added jumpers to BoM https://github.com/sinara-hw/Kasli-SOC/issues/54
- [ ] Added alternate LDOs - TPS74901RGWT
- [ ] Added alternate GPIO expanders - PCA9539APW

gkasprow, 2022-11-23:
@filipswit did you consult moving the EEPROMs to separate I2C interface with software guys?

filipswit, 2022-11-23:
no, but default EEPROM, IC5 is not moved.
We can't add aditional footprint - https://github.com/sinara-hw/Kasli/issues/97, because of conflict with SFP3 addres, so I made same solution as in Kasli v2.0.2

kaolpr, 2022-11-23:
I believe there was some discussion on version naming convention and patch number (according to <major>.<minor>.<patch>) was to be reserved only for compatible component replacements. Here I can see some issue fixes.

If such discussion was not written (honestly, don't remember) maybe we should write down rules on that.

filipswit, 2022-11-23:
@gkasprow and @kaolpr so should I rename it to v1.1?

gkasprow, 2022-11-23:
yes, since it requires software changes, there must be new revision

gkasprow, 2022-11-23:
The R110 and R100 could be replaced with E24 values 
[image]

kaolpr, 2022-11-23:
> yes, since it requires software changes, there must be new revision

@gkasprow How does it require software changes? IO expander has different address / registry map?

gkasprow, 2022-11-23:
Before EEPROM was connected to shared bus, now it has a dedicated FPGA pins.

gkasprow, 2022-11-23:
@filipswit there is yet another [issue ](https://github.com/sinara-hw/EEM_FMC_Carrier/issues/46#issuecomment-1317083831)that needs correction 
The MCP23017 expander port 7 must be output. So we need to swap DISABLE with FAULT
I'd add a reistors for revision ID like in Kasli + one resistor for variant ID (a kind of expanders assembled) placed on the same sheet as I2C expanders

filipswit, 2022-11-23:
>  decouple it from LDOs 1uF caps with a few tens Ohm resistor

You mean to place resistor (22R?) in series with 1uF at all LDOs Vbias inputs?
BTW. I can place any different LDO if you have any other cheap and small.

> The MCP23017 expander port 7 must be output. So we need to swap DISABLE with FAULT
[image]

So it looks that IC24 is fine (these are test points) and in IC25 I should swap 'VUSB_PRESENT_N', with 'SFP_0.LED'?
[image]
> I'd` add a reistors for revision ID like in Kasli + one resistor for variant ID (a kind of expanders assembled) placed on the same sheet as I2C expanders

I'll try to fit 3-4 rev ID bits and one variant ID resistor

gkasprow, 2022-11-23:
> > decouple it from LDOs 1uF caps with a few tens Ohm resistor
> 
> You mean to place resistor (22R?) in series with 1uF at all LDOs Vbias inputs? BTW. I can place any different LDO if you have any other cheap and small.

just add a single 22Ohm (or similar already in the BOM) at the LDO output after the 100nF cap.

gkasprow, 2022-11-23:
> So it looks that IC24 is fine (these are test points) and in IC25 I should swap 'VUSB_PRESENT_N', with 'SFP_0.LED'?

yep, my bad

filipswit, 2022-11-23:
Just verified the DS for this LDO, so it looks 470nF is absolute minimum i this case:
[image]

Would it be good enough solution?
[image]

gkasprow, 2022-11-23:
OK, so you found better DS than me :)

gkasprow, 2022-11-23:
No need for series resistor then.

filipswit, 2022-11-24:
@gkasprow could you please look at the changes one more time?

kaolpr, 2022-12-27:
@gkasprow @filipswit  Can we assume v1.1 is reviewed?

kaolpr, 2022-12-27:
@gkasprow Do you think https://github.com/sinara-hw/Kasli/issues/99 should be implemneted here?

gkasprow, 2022-12-30:
SOT-23 version of EEPROM is still not available

gkasprow, 2022-12-30:
please add note close to the I2C expanders that ASSY variant need to be updated.

gkasprow, 2022-12-30:
why both EEPROMs are not mounted?
[image]

gkasprow, 2022-12-30:
Ah, I see, IC5 is connected instead...
I'd add 24AA025E48T-I/OT assembly variant to it as well.

filipswit, 2022-12-31:
yeep, it should be added. I've made issue to not forget about it in next revision https://github.com/sinara-hw/Kasli-SOC/issues/72.