sinara-hw/Kasli-SOC #74: Undefined behavior with SPI devices on several EEM ports

thomasfire, 2023-05-08:
We've been struggling with the [issue](https://git.m-labs.hk/M-Labs/artiq-zynq/issues/197) that EEM ports 2-7 and 10-11 often fail to work with Samplers, Zotinos, Mirnies and probably some others. 
It appears that these ports are LVDS_18 and not LVDS_25, which are used in RISCV Kasli and other ports on Kasli-SoC. Which leads to the devices fail to recognize the signal because it may be lower than a threshold. Also it seems to be a limitation on Zynq chips https://docs.xilinx.com/v/u/en-US/ug475_7Series_Pkg_Pinout .

Are there any solutions or workarounds about this?

gkasprow, 2023-05-08:
You cannot use LVDS_18 with 2.5V supply on Kasli.
Artix devices support LVDS only on 2.5V ports.
1.8V LVDS ports are supported only in KIntex HP banks.
Kasli SoC banks 12,13 use 2.5V and LVDS25 because these are HR banks
banks 33,34,35 use 1.8V and LVDS18 because these are HP banks

thomasfire, 2023-05-09:
Are there any workarounds/fixes possible? It is quite annoying problem for us

gkasprow, 2023-05-09:
Just use right primitives dedicated to bank types. It is not a bug

gkasprow, 2023-05-30:
I might have an idea of what is going on.
Don't these modules use the same LVDS drivers?
When working with Sayma, I noticed the strange behaviour of 3.3V LVDS transceivers when working with 1.8V banks.
The FPGA bank voltage was elevated to over 2V instead of 1.8V, which was caused by abnormal LVDS transceiver behaviour.
The LVDS transceiver sometimes produced a 3V level on the differential pair when inactive.
When active, the level was around 1.2V which is the desired level for LVDS.

closed 2023-06-02 by thomasfire