sinara-hw/Kasli-SOC #75: EEM8 and EEM9 collide with rails

jordens, 2023-07-19:
EEM8/9 have a small clearance of ~1.5 mm to the board edge. In common board storage systems or crates with 3 mm rail depth this isn't enough and leads to seizing/scratching/damage. If possible move them 0.5 mm, better 1 mm, down. Same for the fan connector if not already done.

kaolpr, 2023-07-20:
Has this issue been previously reported? If not, its severity may be questionable since it has gone unreported for over three years.

Currently, implementing this modification would impede the adoption process of DIOT, as a number of adapters have already been manufactured specifically for this EEM layout.

We can leave this open and if any other layout-altering issues emerge, we can address them in a future release that may require a new version of EEM adapters.

jordens, 2023-07-20:
As stated, it's not an issue in many crates as the rails aren't that deep and there is usually no other rail so close. But in PCB storage systems there is always a rail adjacent.
I don't see how this really hinders DIOT adoption. People will know whether they need the adapter when they order Kasli. Hence there is little desynchronization. Just make a revision in both designs and match the releases/production.
As mentioned previously, also for exactly these reasons I'd rather deprecate this adapter and make a proper Kasli-DIOT. The adapter road especially on Kasli is IMO the biggest blocker in DIOT adoption in Sinara.

linuswck, 2025-06-04:
> EEM8/9 have a small clearance of ~1.5 mm to the board edge. In common board storage systems or crates with 3 mm rail depth this isn't enough and leads to seizing/scratching/damage. If possible move them 0.5 mm, better 1 mm, down. Same for the fan connector if not already done.

To increase the components clearance from board edge, it also requires the FPGA_JTAG connector and some other components to be moved downwards as well. However, there is no physical space to move FPGA_JTAG connector downward. I don't think it can be done without moving and re-routing components around. See the screenshot below. 
[image]

jordens, 2025-06-05:
Best effort would already go a long way. If you can at least move the EEM connectors that would already be a big improvement.

linuswck, 2025-06-05:
I can only move the EEM8/9 connector downward by 1mm. That is what I will do in the new revision #86. 

The fan connector cannot be moved downward due to lack of physical space for the SFP high speed signal. 
See screenshots below
[image]

jordens, 2025-06-06:
You can move it down and left if you rotate it.

jordens, 2025-06-06:
You can also ignore the mounting hole. The only user that I am aware of is the diot adapter which is clearly already moot.

linuswck, 2025-06-06:
@martafafara should I ignore that mounting hole?

martafafara, 2025-06-06:
I'm currently checking all changes, give me some time to consider that

martafafara, 2025-06-06:
I've done the placement study of some solutions and consulted with @kaolpr. The summary is that this mounting hole can be neglected. If we move the EEM connectors down, then the DIOT adapter have to be redesigned and its mounting hole could be also changed. Moreover, this particular mounting point has not been used in any known cases. 
It would require some time, but I found feasible to move also C161 and L18. It could be a big improvement.

martafafara, 2025-06-12:
Yes, remove it, please.