sinara-hw/Kasli-SOC #76: DIOT(CPCI) version of Kasli SOC

gkasprow, 2023-10-11:


jordens, 2023-10-11:
Let's do this.
Could you propose a rough block diagram or schematic sketch that we can iterate on?
Is there no OHWR/CERN DIOT system board with a SoC that we could align with?

marmeladapk, 2023-10-11:
We'll probably take a lot from DIOT System Board (
https://ohwr.org/project/diot-sb-zu/).

gkasprow, 2023-10-12:
We will only replace EEM connectors with CPCIS ones and use bigger FPGA. Rest remains the same.
An important feature is correct power sequencing that should improve Kasli's survivability.

jordens, 2023-10-26:
Having reviewed the DIOT sys board it looks like we can take it as is.

jordens, 2023-11-08:
... especially if the planned changes for v3 land (RJ45 to PS, USB for JTAG/I2C etc).