sinara-hw/Kasli-SOC #77: ftdi serial number

jbqubit, 2023-10-13:
The USB-serial interface chip is a FT4232H QUAD HIGH SPEED USB TO MULTIPURPOSE UART. This chip is accompanied by an EEPROM on Kasli-SOC. FTDI supports assignment of a factory-defined serial number to the chip which stored on the EEPROM. This is set using the FTDI tool [FT_PROG](https://ftdichip.com/utilities/). 

My Kasli-SOC shipped from Technosystem with no flashed serial number. This is problematic because Kasli-SOC can't be deterministically identified in a system without it. 

```
$ python3 -m serial.tools.list_ports -v
/dev/ttyS0          
    desc: ttyS0
    hwid: PNP0501
/dev/ttyUSB3        
    desc: Quad RS232-HS - Quad RS232-HS
    hwid: USB VID:PID=0403:6011 LOCATION=1-14.4:1.1
/dev/ttyUSB4        
    desc: Quad RS232-HS - Quad RS232-HS
    hwid: USB VID:PID=0403:6011 LOCATION=1-14.4:1.2
/dev/ttyUSB5        
    desc: Quad RS232-HS - Quad RS232-HS
    hwid: USB VID:PID=0403:6011 LOCATION=1-14.4:1.3
5 ports found
```

After using FT_PROG...
```
$ python3 -m serial.tools.list_ports -v
/dev/ttyS0          
    desc: ttyS0
    hwid: PNP0501
/dev/ttyUSB0        
    desc: FT232R USB UART - FT232R USB UART
    hwid: USB VID:PID=0403:6001 SER=AL02CA9T LOCATION=1-14.3.1
/dev/ttyUSB1        
    desc: USB-serial - USB-serial
    hwid: USB VID:PID=0403:6011 SER=umd-ksoc-001 LOCATION=1-14.4:1.0
/dev/ttyUSB2        
    desc: USB-serial - USB-serial
    hwid: USB VID:PID=0403:6011 SER=umd-ksoc-001 LOCATION=1-14.4:1.1
/dev/ttyUSB3        
    desc: USB-serial - USB-serial
    hwid: USB VID:PID=0403:6011 SER=umd-ksoc-001 LOCATION=1-14.4:1.2
/dev/ttyUSB4        
    desc: USB-serial - USB-serial
    hwid: USB VID:PID=0403:6011 SER=umd-ksoc-001 LOCATION=1-14.4:1.3
6 ports found
```

Please add EEPROM s/n programming to the Kasli-SOC factory test suite.

jbqubit, 2023-10-13:
Also, the Product Description field on the EEPROM is a generic "Quad RS232-HS". This could be something more descriptive such as "sinara kasli-soc". This would make it easier for users to identify the board and would require a trivial change to the openocd script ```kasli_soc.cfg```.

marmeladapk, 2023-10-13:
AFAICT CTI does this, Quartiq and M-labs probably as well with https://github.com/quartiq/kasli-i2c . Please direct this to manufacturers directly, as there's nothing to do in this PCB project. 

Also changing name would break compatibility of openocd with older boards, but you can discuss this in m-labs fork of openocd or in artiq repository.

closed 2023-10-13 by marmeladapk

jbqubit, 2023-10-16:
In case somebody looks at this Issue in the future I'll summarize. [Most ARTIQ infrastructure](https://git.m-labs.hk/M-Labs/zynq-rs/src/commit/c15b54f92b3d4e125ae47a0dce7abe4b2bc9e054/openocd/kasli_soc.cfg) assumes that Kasli boards have the ```ftdi_device_desc``` field stuffed with ```Quad RS232-HS```. Unique identification requires that the each Kasli SOC have a unique serial number -- the [FT_PROG](https://ftdichip.com/utilities/) tool can do this automatically.