sinara-hw/Kasli-SOC #81: 12V input short circult

IamTaoChen, 2024-06-04:
It seems that my Kasli-SOC was short circuit and the fan doesn't work.  The chip `LTM4644IY` is a little hot.

[image]

gkasprow, 2024-06-04:
what do you mean by "was short circuit"?
Did you short any pins?

IamTaoChen, 2024-06-05:
I'm not sure.
I wanted to install the firmware into the SD card. I didn't install it back on the rack when I powered it on,  I held the card, and maybe, the back touched something.

gkasprow, 2024-06-05:
Well, the SD slot was not designed for life insertion.
How can I help you? You should send the PCB to the manufacturer for repair then.

IamTaoChen, 2024-06-05:
I just want to know how to do a basic test to know which part is burned.

gkasprow, 2024-06-05:
critical supplies are missing so it's either SMPS or FPGA which is broken