sinara-hw/Kasli-SOC #88: v1.2rc2 changelog

martafafara, 2025-08-28:
- [ ] most changes to v1.2 described in [pull request](https://github.com/sinara-hw/Kasli-SOC/pull/86)
- [ ] introduce #84 
- [ ] introduce #87 
- [ ] following the approval of @gkasprow in the [discussion](https://github.com/sinara-hw/meta/issues/95) Amphenol 132203 will be replaced by Delock 65848

filipswit, 2025-09-03:
Why there is v1.2.0rc2 on overlay? Shouldn't it be just v1.2? or v1.2.0 ? is it final version or still under development?

[image]

martafafara, 2025-09-08:
Above mentioned remarks introduced. Corrected also #89 and #90. Release created.

martafafara, 2025-09-22:
#92 introduced

atseml, 2025-09-24:
@maciejprzybysz The provided source code zip and tarball files in the [v1.2 release](https://github.com/sinara-hw/Kasli-SOC/releases/tag/v1.2) are out of sync with the released PDFs. The files and the "v1.2" tag are still at v1.2rc3.